Universal-Debloater-Alliance/universal-android-debloater-next-generation #711: Update app description of pkg(com.samsung.app.telephoneyui):

SchneeSchmitt, 2024-11-06:
### List your packages

- Package name: com.samsung.app.telephoneyui
- Current list: Advanced, OEM
"Call settings"

- Current description:
It's calling app from samsung China, Korea.
I don't know how important this app is.

- Proposed description:
Lets you access the calling settings including gestures, blocked numbers and hide caller id.
Some third-party dialers like Google Phone also have settings forward to this package.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2024-11-25:
@SchneeSchmitt Can you confirm that removing only that package in fact breaks the things mentioned? I cannot reproduce this, telephoneyui is in-call UI, there are other packages that control the settings mentioned.

Manolo5678, 2025-02-21:
Hi! Disabling this package broke the following options on my S20+ 5G:

Settings -> Connections

- SIM Manager
`It is displayed but does nothing when pressed`

- Mobile Networks
`It does not appear, making it impossible to access certain settings such as Data Roaming, VoLTE calls, Network Mode, APN settings, and Network Operators settings`

PR #978 (2025-06-06): pkg(scope): Issues with SIM Manager, Package Removal, and Performance
Fixes #792
Fixes #711
commit 10bdbd30 (2025-06-06): pkg(scope): Issues with SIM Manager, Package Removal, and Performance
Fixes #792
Fixes #711
closed 2025-03-22 by AnonymousWP