Universal-Debloater-Alliance/universal-android-debloater-next-generation #713: removing com.android.carrierdefaultapp breaks data connection on oppo device

Asra05, 2024-11-09:
### Describe the problem

H

### You have a solution?

The solution is to reinstall the package to restore mobile data functionality.

### Provide logs

J

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

SchneeSchmitt, 2024-11-12:
Data still works fine for my OnePlus Ace 3V on ColorOS 14 and 15
Although it's quite normal to see a package breaks something on specific devices

schbrns, 2024-11-25:
The description already mentions this exact breakage.

Please feel free to re-open the issue if you have any other questions. Thanks.

closed 2024-11-25 by schbrns