Universal-Debloater-Alliance/universal-android-debloater-next-generation #718: pkg(scope): force close on Ruby if I use recommended apps on A14 CN firmware

rayman9595, 2024-11-16:
### Describe the problem

Hi..if I use recommended uninstall, I got Force close when I swipe down on the upper screen to get control center..it come back on screen...I restore everything and no issue..I don't search to find the culprit..

### Provide logs

2024-11-16 02:29:44 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:29:44 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:29:55 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:29:55 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:30:58 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:30:58 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:31:11 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-11-16 02:31:11 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 34 | DEVICE: Redmi 22101316G
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 02:31:12 WARN  [src\core\uad_lists.rs:213] Could not load remote debloat list: https://raw.githubusercontent.com/Universal-Debloater-Alliance/universal-android-debloater/main/resources/assets/uad_lists.json: Connection Failed: Connect error: Une tentative d’accès à un socket de manière interdite par ses autorisations d’accès a été tentée. (os error 10013)
2024-11-16 02:31:13 WARN  [src\core\uad_lists.rs:213] Could not load remote debloat list: https://raw.githubusercontent.com/Universal-Debloater-Alliance/universal-android-debloater/main/resources/assets/uad_lists.json: Connection Failed: Connect error: Une tentative d’accès à un socket de manière interdite par ses autorisations d’accès a été tentée. (os error 10013)
2024-11-16 02:31:14 ERROR [src\gui\views\list.rs:878] Error loading remote debloat list for the phone. Fallback to embedded (and outdated) list
2024-11-16 02:31:21 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 34 | DEVICE: Redmi 22101316G
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.bluetoothmidiservice -> Package com.android.bluetoothmidiservice installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.cameraextensions -> Package com.android.cameraextensions installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.calllogbackup -> Package com.android.calllogbackup installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.inputsettings.overlay.miui -> Package com.android.inputsettings.overlay.miui installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.federatedcompute.services -> Package com.android.federatedcompute.services installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.microdroid.empty_payload -> Package com.android.microdroid.empty_payload installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.overlay.gmssettingprovider -> Package com.android.overlay.gmssettingprovider installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.overlay.gmscontactprovider -> Package com.android.overlay.gmscontactprovider installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.settings.overlay.miui -> Package com.android.settings.overlay.miui installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.role.notes.enabled -> Package com.android.role.notes.enabled installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.stk.overlay.miui -> Package com.android.stk.overlay.miui installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.health.connect.backuprestore -> Package com.android.health.connect.backuprestore installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.egg -> Package com.android.egg installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.adservices.api -> Package com.android.adservices.api installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 cn.wps.moffice_eng.xiaomi.lite -> Package cn.wps.moffice_eng.xiaomi.lite installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.systemui.overlay.miui -> Package com.android.systemui.overlay.miui installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.theme.font.notoserifsource -> Package com.android.theme.font.notoserifsource installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.hotwordenrollment.xgoogle -> Package com.android.hotwordenrollment.xgoogle installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.healthconnect.controller -> Package com.android.healthconnect.controller installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.nearby.halfsheet -> Package com.android.nearby.halfsheet installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.hotwordenrollment.okgoogle -> Package com.android.hotwordenrollment.okgoogle installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.debug.loggerui -> Package com.debug.loggerui installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.goodix.gftest -> Package com.goodix.gftest installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.fingerprints.sensortesttool -> Package com.fingerprints.sensortesttool installed for user: 0
2024-11-16 02:32:06 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.ondevicepersonalization.services -> Package com.android.ondevicepersonalization.services installed for user: 0
2024-11-16 02:32:07 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.google.android.gms.location.history -> Package com.google.android.gms.location.history installed for user: 0
2024-11-16 02:32:07 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.google.android.feedback -> Package com.google.android.feedback installed for user: 0
2024-11-16 02:32:07 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.google.android.overlay.modules.captiveportallogin.forframework -> Package com.google.android.overlay.modules.captiveportallogin.forframework installed for user: 0
2024-11-16 02:32:07 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.google.android.overlay.modules.documentsui -> Package com.google.android.overlay.modules.documentsui installed for user: 0
[... cut ...]
2024-11-16 02:36:34 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 02:36:38 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-11-16 02:37:20 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-11-16 02:37:54 ERROR [src\core\utils.rs:99] Can't open the following URL: 

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

rayman9595, 2024-11-16:
no more error if I don't select all and if I keep some MTK and MI apps...

2024-11-16 02:29:44 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:29:44 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:29:55 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:29:55 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:30:58 ERROR [src\core\sync.rs:72] ADB: program not found
2024-11-16 02:30:58 ERROR [src\core\sync.rs:73] ADB was not found
2024-11-16 02:31:11 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-11-16 02:31:11 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 34 | DEVICE: Redmi 22101316G
2024-11-16 02:31:12 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 02:31:12 WARN  [src\core\uad_lists.rs:213] Could not load remote debloat list: https://raw.githubusercontent.com/Universal-Debloater-Alliance/universal-android-debloater/main/resources/assets/uad_lists.json: Connection Failed: Connect error: Une tentative d’accès à un socket de manière interdite par ses autorisations d’accès a été tentée. (os error 10013)
2024-11-16 02:31:13 WARN  [src\core\uad_lists.rs:213] Could not load remote debloat list: https://raw.githubusercontent.com/Universal-Debloater-Alliance/universal-android-debloater/main/resources/assets/uad_lists.json: Connection Failed: Connect error: Une tentative d’accès à un socket de manière interdite par ses autorisations d’accès a été tentée. (os error 10013)
2024-11-16 02:31:14 ERROR [src\gui\views\list.rs:878] Error loading remote debloat list for the phone. Fallback to embedded (and outdated) list
2024-11-16 02:31:21 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 34 | DEVICE: Redmi 22101316G
2024-11-16 02:31:22 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.bluetoothmidiservice -> Package com.android.bluetoothmidiservice installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.cameraextensions -> Package com.android.cameraextensions installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.calllogbackup -> Package com.android.calllogbackup installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.inputsettings.overlay.miui -> Package com.android.inputsettings.overlay.miui installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.federatedcompute.services -> Package com.android.federatedcompute.services installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.microdroid.empty_payload -> Package com.android.microdroid.empty_payload installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.overlay.gmssettingprovider -> Package com.android.overlay.gmssettingprovider installed for user: 0
2024-11-16 02:32:02 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.overlay.gmscontactprovider -> Package com.android.overlay.gmscontactprovider installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.settings.overlay.miui -> Package com.android.settings.overlay.miui installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.role.notes.enabled -> Package com.android.role.notes.enabled installed for user: 0
2024-11-16 02:32:03 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.stk.overlay.miui -> Package com.android.stk.overlay.miui installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.health.connect.backuprestore -> Package com.android.health.connect.backuprestore installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.egg -> Package com.android.egg installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.adservices.api -> Package com.android.adservices.api installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 cn.wps.moffice_eng.xiaomi.lite -> Package cn.wps.moffice_eng.xiaomi.lite installed for user: 0
2024-11-16 02:32:05 INFO  [src\core\sync.rs:131] [Recommended] cmd package install-existing --user 0 com.android.systemui.overlay.miui -> Package com.android.systemui.overlay.miui installed for user: 0
[... cut ...]
2024-11-16 12:38:52 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2024-11-16 12:39:58 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2024-11-16 12:39:58 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-11-16 12:39:58 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2024-11-16 12:39:58 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found

AnonymousWP, 2025-02-02:
Really looks like an internet issue, because it can't fetch the URL. Should work fine.

closed 2025-02-02 by AnonymousWP