Universal-Debloater-Alliance/universal-android-debloater-next-generation #722: bug(scope): Latest Release & Artifact Not Working On MacOS Ventura Intel

alsyundawy, 2024-11-17:
### Describe the bug

bug(scope): Latest Release & Artifact Not Working On MacOS Ventura Intel

### Expected behavior

bug(scope): Latest Release & Artifact Not Working On MacOS Ventura Intel

### You have a solution?

bug(scope): Latest Release & Artifact Not Working On MacOS Ventura Intel

### Operating System

MacOS

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Frigyes06, 2024-11-17:
What do you mean by not working? Can you post logs please?

AnonymousWP, 2024-11-25:
Yes you're completely right, I'll add it to the milestone of 1.1.1. This should have been issue even before 1.1.0 though. Do you recall that?

Frigyes06, 2024-11-26:
So apparently macos-latest got changed to the M1 machines, as opposed to intel:
https://docs.github.com/en/actions/using-github-hosted-runners/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources

We will need to add another run on macos-13

alsyundawy, 2024-12-09:
1.0.3 no self update running well on monterey intel

[image]

supersophie, 2024-12-28:
Hi,

What is the plan for Intel macos releases?

supersophie, 2025-01-06:
As in some people have got macs running on Intel, and it would really great if we could run this.

closed 2025-01-12 by AnonymousWP

AnonymousWP, 2025-01-13:
@alsyundawy @mallardtheduck @supersophie First: sorry that it had to take this long to get fixed.

It should be fixed in these 2 PRs:

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/791
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/793

Could you try it out? Here's the link to the newly generated Intel-artifact: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12746391399/artifacts/2421626322

alsyundawy, 2025-01-13:
working like charm, tested. Thx for ward work, nice release

[image]

AnonymousWP, 2025-01-13:
> working like charm, tested. Thx for ward work, nice release
> 
> <img alt="image" width="958" src="https://private-user-images.githubusercontent.com/6451849/402722338-606103dc-c407-4bd3-a580-1eb7dc77e88f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY4MDg2NzQsIm5iZiI6MTczNjgwODM3NCwicGF0aCI6Ii82NDUxODQ5LzQwMjcyMjMzOC02MDYxMDNkYy1jNDA3LTRiZDMtYTU4MC0xZWI3ZGM3N2U4OGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTNUMjI0NjE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2Y1NzMzZTgwN2ZjZWM2ODQxYjVkMjg0ZTUyNmRmOTgxYjAyYmI2ODA5MjRiMDE2YTFlODZmYThhZjMyNGQ1OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.qsReNU6CqoORKnYtP2_Zz_G440975UYJykZvg4cj3BE">

Great to hear, thanks!

alsyundawy, 2025-01-16:
hai @AnonymousWP can you release arm & intel on release

thx for hard work

AnonymousWP, 2025-02-04:
We will once we finish 1.1.1: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/milestone/6