Universal-Debloater-Alliance/universal-android-debloater-next-generation #724: bug(scope): When trying to uninstall spay framework  I get this error.

SJ1O, 2024-11-21:
### Describe the bug

When trying to uninstall spay framework I get this error.
Failed to perform ADB operation

[Recommended] pm uninstall-user 0 com.samsung.android.spayfw-> Failure [DELETE_FAILED_OWNER_BLOCKED]

### Expected behavior

Uninstall app

### Operating System

Windows

### Distro version

Windows 11

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-11-21:
It means Samsung doesn't allow you to uninstall the app. Try disabling instead.

closed 2024-11-21 by AnonymousWP

SJ1O, 2024-11-21:
Is the only way to disabled on my phone or can I use uadb?

AnonymousWP, 2024-11-22:
> Is the only way to disabled on my phone or can I use uadb?

You can either use UAD-ng or ADB to disable an app. Sometimes, Samsung allows you to disable the app from the settings of OneUI.

SJ1O, 2024-11-22:
> > Is the only way to disabled on my phone or can I use uadb?
> 
> You can either use UAD-ng or ADB to disable an app. Sometimes, Samsung allows you to disable the app from the settings of OneUI.

Could you tell me how to disable using uad-ng please. I looked in the settings and cannot find a disabled option only uninstall. I did read the faq but it doesn't explain how to use the disable option.

SJ1O, 2024-11-22:
> > Is the only way to disabled on my phone or can I use uadb?
> 
> You can either use UAD-ng or ADB to disable an app. Sometimes, Samsung allows you to disable the app from the settings of OneUI.

Nevermind I found the settings to disable apps I wasn't looking very good. Thanks for all your work and app