Universal-Debloater-Alliance/universal-android-debloater-next-generation #732: Samsung Intelligence Voice Services change

YoshiiShell, 2024-11-24:
### List your packages

- Package name: com.samsung.android.intellivoiceservice
- Current list: Unlisted
  - Proposed list: oem
- Current removal: Unlisted
  - Proposed removal: Recommended
- Current description: "CONTRIBUTION WELCOMED"
  - Proposed description: Appears to be an AI speech recognition service. Does not look like it would break anything maybe other than Samsung voice detection services like transcripts in Samsung Voice Recorder.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-11-25:
For anyone skeptical, there are [screenshots in the `#general` Discord channel](https://discord.com/channels/1168607797081022514/1168607798196715573/1310337869042286742). Here's a highlight:
[image]

AnonymousWP, 2024-11-25:
> For anyone skeptical, there are [screenshots in the `#general` Discord channel](https://discord.com/channels/1168607797081022514/1168607798196715573/1310337869042286742). Here's a highlight: [image]

Your image URL is dead. 😉

Rudxain, 2024-11-29:
> Your image URL is dead. 😉

I hate Discord :smiling_face_with_tear:, so unreliable

closed 2025-03-22 by AnonymousWP