Universal-Debloater-Alliance/universal-android-debloater-next-generation #746: feat: CLI

Rudxain, 2024-11-28:
### Describe the feature you want

## What
Quoting [the chat](https://discord.com/channels/1168607797081022514/1171387180409692160/1310831109461774438) with minor edits:
> I've been considering to develop **"UAD-NG: CLI edition"** (the name can be shortened to "UAD-CLI", as the legacy UAD didn't have a command-line interface).
>
> Thanks to [`clap`](https://crates.io/crates/clap), this could be easy. I could even implement #184 while we're at it
>
> I don't have time for this, but I wish to see it happen.
>
> `clap` may conflict with `iced`, and the CLI is supposed to be "lite"

## How
I thought a separate repo (on my account, and then later on the org) was a good idea, but @schbrns suggested that it might better to take the [mono-repo](https://en.wikipedia.org/wiki/Monorepo) approach, as there's some internal code that can be shared between GUI and CLI (such as the handling of ADB commands, and the logic for different Android versions).

It's possible to convert to a mono-repo, there are 3 ways:
- Multi-crate, which is what [`b3sum`](https://github.com/BLAKE3-team/BLAKE3) does.
- Single-crate multi-pack, via [workspace](https://doc.rust-lang.org/book/ch14-03-cargo-workspaces.html), which is what [`hx`](https://github.com/helix-editor/helix) does.
- Single-crate single-pack, [with a big caveat](https://github.com/rust-lang/cargo/issues/1982)

It seems workspaces are the most sensible option, as multi-crate would require us to have 3 crates (is this 100% true?):
- core library
- GUI importing core
- CLI importing core

## Why?
CLIs tend to be more accessible (#605) and reliable (#645) than GUIs. This would allow more potential users to use the app with minimal problems

The CLI can be designed to be more similar to a typical package manager, which could make it more intuitive and safe (#642) to use than the GUI.

Also, the CLI wouldn't need as much logging as the GUI (shells write to a `.*_history` file), which would make it even more efficient.

> why not [TUI](https://en.wikipedia.org/wiki/Text-based_user_interface)?

Because that's just a GUI made of text, it's a waste of time. CLI and GUI complement each other.

## Drawbacks
If we do design it as a typical pack-man, we would need to somehow cache the list of installed packs from every connected device, as a command such as `uad ls -i` (list installed packs, rather than the pack-IDs in `uad_lists.json`) could take **hundreds of milliseconds** to sync.

Rather than caching to storage, a [REPL](https://en.wikipedia.org/wiki/Read%E2%80%93eval%E2%80%93print_loop) might be a better choice, as the state of the session is always in memory.

## etc...
We may have to revert #154

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-12-02:
I've just realized another benefit of the CLI. If the CLI and GUI are separate executables, then users could run it on Android itself!

> [!note]
> In theory, this is **already possible for the GUI** too:
> - [x11 apps on Termux](https://wiki.termux.com/wiki/Graphical_Environment) (needs add-ons)
> - [Iced apps on Android](https://github.com/iced-rs/iced/discussions/859#discussioncomment-691719)
> I haven't tested either of those

This can happen in 3 ways:
- The users [build it themselves](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Building-from-source/80b64fd8eb01619d29cf3ccab6e403221aa4e6c5); as [Rs](https://github.com/termux/termux-packages/tree/bd5a9f93c001677881220146ef149aad12764b21/packages/rust), [Clang](https://github.com/termux/termux-packages/blob/bd5a9f93c001677881220146ef149aad12764b21/packages/libllvm/clang.subpackage.sh), and (surprisingly) even [Mold](https://github.com/termux/termux-packages/tree/master/packages/mold) are available in the `Main` repo!
- We release binaries statically-linked with [`musl`](https://rust-lang.github.io/rustup-components-history/aarch64-unknown-linux-musl.html)
- We release binaries dynamically-linked to [Bionic](https://en.wikipedia.org/wiki/Bionic_(software)) via [NDK](https://developer.android.com/ndk) (see [this](https://rust-lang.github.io/rustup-components-history/aarch64-linux-android.html) and [this](https://doc.rust-lang.org/rustc/platform-support/android.html))

> [!tip]
> There's also [`cross`](https://github.com/cross-rs/cross)

However, it's not recommended to disable packs on the same device, as the user could permanently lock themselves out, requiring a factory-reset

Iamanaws, 2025-10-26:
I think it would be a good idea to add a `comment` or `summary` attribute to pkgs, so we can have a short / one-liner note and expand on description if required.

This is a common field on some package managers and could make things cleaner, also would be useful for a CLI when listing and searching pkgs

Rudxain, 2025-11-03:
> `comment` or `summary`

I agree, but that would be more useful if descriptions were longer, which could happen in the future if we [convert the JSON into multiple markdown files](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/608#discussioncomment-10643750)