Universal-Debloater-Alliance/universal-android-debloater-next-generation #747: pkg(com.mi.healthglobal): it also disable the heart beats meter via on-screen fingerprint reader

kuolemaaa, 2024-11-29:
### List your packages

- Package name: com.mi.healthglobal
- Current list: oem
- Current removal: Recommended
- Current description:
>Hi Health
Pedometer, menstrual and sleep tracker
[...]
  - Proposed description: 
>Hi Health
Pedometer, heartbeat rate meter, menstrual and sleep tracker
[...]

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2025-01-02:
https://www.xda-developers.com/mi-health-xiaomi-fitness-app/

kuolemaaa, 2025-01-03:
> https://www.xda-developers.com/mi-health-xiaomi-fitness-app/

what?

i'm talking about https://www.youtube.com/watch?v=7oXPky_XFsc

schbrns, 2025-01-04:
Sorry, posted the wrong link, same package though.

https://www.xda-developers.com/xiaomi-mi-health-app-gets-heart-rate-monitoring-support-using-camera/

closed 2025-02-06 by AnonymousWP