Universal-Debloater-Alliance/universal-android-debloater-next-generation #758: Adding description of pkg(com.oplus.pscanvas):

SchneeSchmitt, 2024-12-15:
### List your packages

- Package name: com.oplus.pscanvas
- List: OEM
- Removal: Advanced
- Description: 

Removing this package breaks split screen.

### Acknowledgements

- [ ] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

SchneeSchmitt, 2024-12-15:
Also quite funny that this package does improve the stability of the launcher animations. I can keep triggering the disturbing animations faster without failed with this package installed.

schbrns, 2025-01-02:
https://community.oneplus.com/thread/1484307668276346883

Leaving this here to leave package description from later.

AnonymousWP, 2025-02-03:
Thanks, changed the removal to `Expert` though as I can see how useful it is for OnePlus Pad (2)/OnePlus Open users.

closed 2025-02-06 by AnonymousWP