Universal-Debloater-Alliance/universal-android-debloater-next-generation #761: Multiple packages from TCL 8182 tablet (Android 8.1.0)

Lanchon, 2024-12-20:
### List your packages

- Package name: com.facebook.lite
- List: Misc
- Removal: Recommended
- Description: The Facebook Lite app.

- Package name: com.android.store
- List: OEM
- Removal: Recommended
- Description: An app called TCL Mobile that you cannot disable and just opens up a link to the OEM in the browser.

- Package name: com.cide.interactive.tclcontrolparental
- List: Misc
- Removal: Advanced
- Description: TCL version of Parental Center, a parental control app.

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-12-20:
Are you sure the Facebook Lite app is not a user app? Meaning you can't uninstall it via the settings of your tablet.

Lanchon, 2024-12-21:
@AnonymousWP,

> Are you sure the Facebook Lite app is not a user app?

yes i am. you can disable it from its "app info" settings page, but you cannot uninstall it.

and btw...

before you ask, the weird package name `com.android.store` mentioned above is correct.

closed 2025-02-06 by AnonymousWP