Universal-Debloater-Alliance/universal-android-debloater-next-generation #762: pkg(scope): com.samsung.android.location breaks photo location tags

SJ1O, 2024-12-22:
### List your packages

- Package name:  com.samsung.android.location
- Current list: recommended 
  - Proposed list: expert?
- Current removal: 
  - Proposed removal: 
- Current description: not on my laptop but it doesn't mention breaking location tagging of photos
  - Proposed description: add that it breaks location tagging on photos?

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

wwqwq2313, 2024-12-22:
Hello, we tried to solve the issue.

This is what we did:

Changed the list from 'recommended' to 'expert' for com.samsung.android.location package. Updated the description to mention that removing this package breaks location tagging on photos.

You can review changes in this commit: https://github.com/wwqwq2313/Universal-Debloater-Alliance-universal-android-debloater-next-generation-762/commit/bdd0b68bff58edd1ecff99a20ab3762d35510e8c.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

Rudxain, 2025-01-01:
Thanks for the report! I was curious to know the purpose of that pack.

I'd say it's `Advanced` removal

Rudxain, 2025-01-01:
Can you please specify the device model you tested? And by "tagging" you mean that the Camera app can't add location metadata to taken photos, or that the Gallery can't decode them? (maybe both?)

SJ1O, 2025-01-01:
> Can you please specify the device model you tested? And by "tagging" you mean that the Camera app can't add location metadata to taken photos, or that the Gallery can't decode them? (maybe both?)

Hi, I tested it on a Samsung galaxy s24 base model. I uninstalled the gallery app and use Google photos so I'm not sure about that part. However yes there is no location meta data when checking the photo via Google photos not sure about the gallery app but I suspect it would be the same. I had to restart the device to notice this issue / resolve the issue. I tested it several times and definitely narrowed it down to this package.

schbrns, 2025-01-02:
Is this actual need-to-know information?

If you're removing the Samsung Location package, isn't it assumed that any Samsung component with the word "Location" in it will also be removed/broken? In this case, "**Location** tags".

From a logical and privacy standpoint, I'll maintain that this belongs in Recommended.
From a functionality standpoint, for those wanting location tags, I can maybe see the argument for Advanced.

But there are two practises going against each other here.

SJ1O, 2025-01-02:
> Is this actual need-to-know information?
>
> If you're removing the Samsung Location package, isn't it assumed that any Samsung component with the word "Location" in it will also be removed/broken? In [...]

Assumptions are made and most assumptions are wrong. I think any information is handy. If the universal debloater app didn't have decent information about packages like it currently does I wouldn't have removed over 200 apps. The more information the better understanding we have of what we should remove. I don't think this package broke anything else only location meta data from my testing. GPS and everything else worked fine.

SJ1O, 2025-01-02:
> Thanks for the report! I was curious to know the purpose of that pack.
> 
> I'd say it's `Advanced` removal. Edit: It's already `Advanced` 😕

Sorry I wasn't on my laptop I thought it was in recommended for some reason 😬. However I think the information will come in handy for people looking to remove that package.

Rudxain, 2025-01-03:
> isn't it assumed that any Samsung component with the word "Location" in it will also be removed/broken?

That's exactly why I removed the `package::breakage` label. The docs already imply this kind of breakage is to be expected. But they need to be improved anyway, so I opened the PR.

> Sorry I wasn't on my laptop I thought it was in recommended for some reason 😬

oh ok, lol

PR #773 (2025-01-06): pkg(samsung): improve S-location related descriptions
- closes #762
- Updates descriptions of:
  - `com.samsung.android.location`: add breakage example and display-name.
  - `com.sec.location.nsflp2`
  - `com.samsung.locationhistory`: grammar

I want to test some Galaxy phones before merging this, to see how widespread this issue is
closed 2025-01-06 by AnonymousWP