Universal-Debloater-Alliance/universal-android-debloater-next-generation #764: bug(scope): Package description content isn't cleaned after uninstall

santo998, 2024-12-23:
### Describe the bug

After uninstalling a package, it gets deleted from the list, but it's description isn't cleaned.

### Expected behavior

After uninstalling a package, description should be empty, because no package is selected.

### Operating System

Windows

### Distro version

Win 11 Pro

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2025-01-04:
Personally, this is useful behavior for me. It would be more intuitive if the pack [stayed visible in the list](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/642#issuecomment-2369661188), then there would be no need to clear the description box

AnonymousWP, 2025-01-06:
Could you include a screen recording or some screenshots? I'm pretty sure I understand what you mean here, but just to be sure.

Rudxain, 2026-10-24:
Closing as stale

closed 2026-10-24 by Rudxain