Universal-Debloater-Alliance/universal-android-debloater-next-generation #767: bug(scope): Bad CPU on Macos 12 when running uad-ng-macos

supersophie, 2024-12-26:
### Describe the bug

Hi,

I downloaded your uag binary, but got this:

```
$ ./uad-ng-macos 
-bash: ./uad-ng-macos: Bad CPU type in executable:~/Downloads sophie$ file uad-ng-macos
uad-ng-macos: Mach-O 64-bit executable arm64
$ uname -a
Darwin blah 21.6.0 Darwin Kernel Version 21.6.0: Mon Jun 24 00:56:10 PDT 2024; root:xnu-8020.240.18.709.2~1/RELEASE_X86_64 x86_64
```

Have you got one for the Macbooks running macos 12?

```
ADB version: $ adb --version
Android Debug Bridge version 1.0.41
Version 35.0.2-12147458
Installed as /usr/local/bin/adb
Running on Darwin 21.6.0 (x86_64)

```

Regards.

### Expected behavior

The binary would execute on macos.

### You have a solution?

"Have you got a solution?" 

No, I have not. 

### Operating System

MacOS

### Distro version

macos 12.7.6

### ADB version

1.0.41

### Provide logs

<details><summary>Log</summary>

```
$ ./uad-ng-macos 
-bash: ./uad-ng-macos: Bad CPU type in executable:~/Downloads sophie$ file uad-ng-macos
uad-ng-macos: Mach-O 64-bit executable arm64
$ uname -a
Darwin blah 21.6.0 Darwin Kernel Version 21.6.0: Mon Jun 24 00:56:10 PDT 2024; root:xnu-8020.240.18.709.2~1/RELEASE_X86_64 x86_64
```
</details>
$ ./uad-ng-macos 
-bash: ./uad-ng-macos: Bad CPU type in executable:~/Downloads sophie$ file uad-ng-macos
uad-ng-macos: Mach-O 64-bit executable arm64
$ uname -a
Darwin blah 21.6.0 Darwin Kernel Version 21.6.0: Mon Jun 24 00:56:10 PDT 2024; root:xnu-8020.240.18.709.2~1/RELEASE_X86_64 x86_64

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

NitwitVillager, 2024-12-27:
The same thing happened to me too on macOS sequoia (15.2)
[image]

Frigyes06, 2024-12-28:
Duplicate of #722

closed 2024-12-28 by Frigyes06