Universal-Debloater-Alliance/universal-android-debloater-next-generation #770: feat(removal): Treat system UIDs as `unsafe`

Rudxain, 2024-12-31:
### Describe the feature you want

If any package belongs to any of these special Linux user IDs:
- `0`: `root`
- `1000`: `system`
- `2000`: `shell`

Then UADNG should refuse to disable them, as if they were marked as `unsafe`.

This seems necessary, because the actual pack-IDs that belong to these high-privilege groups can vary across devices, so we can't simply mark them all as `expert` or `unsafe`.

How to implement:
- https://developer.android.com/tools/dumpsys#uid_stats
- https://stackoverflow.com/questions/39077820/find-package-name-of-android-app-by-uid-without-root-permissions/39077932#39077932

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

schbrns, 2025-01-01:
Further reading:
https://developer.android.com/reference/android/app/AppOpsManager

> App-ops can either be controlled for each uid or for each package. Which one is used depends on the API provider maintaining this app-op. For any security or privacy related app-op the provider needs to control the app-op for per uid as all security and privacy is based on uid in Android.

Galaxy S23, Android 14, One UI 6.1.1
I've thoroughly tested with adb and Proton VPN split tunneling: 
EVERY uid 1000 app needs to be enabled AND have full internet access to run system updates.

You'll notice this is why apps like RethinkDNS pre-merge the UIDs of apps, e.g. if you modify one UID 1000 app, you modify all of them the same way.

Rudxain, 2025-01-01:
> apps like RethinkDNS pre-merge the UIDs

True, NetGuard does it too.

The funny thing is that the built-in system settings app (on every device I tried) doesn't immediately update the perms for an app group; we have to "reload" the list to see the new perms for all apps in a group. Bad UX at its finest

schbrns, 2025-01-02:
> we have to "reload" the list to see the new perms
Goes for more than just new perms, this happens in many cases in Settings > Apps.

> Bad UX at its finest
Truly.

Rudxain, 2025-01-04:
[Found more info](https://stackoverflow.com/a/8685751). I already added the official doc link to the initial comment