Universal-Debloater-Alliance/universal-android-debloater-next-generation #772: bug(scope): As has happened to others, the program does not open on windows 10 and nothing is created in the uad folder

bobcov, 2025-01-01:
### Describe the bug

Not sure what to look for.  ADB works for other programs and at the command prompt.  When I launch the debloater, nothing happens. If I delete the uad folder and launch again, the folder is recreated, but there are no log files and no gui. 

### Expected behavior

The program should open.

### You have a solution?

No solution

### Operating System

Windows

### Distro version

Win 10 22H2

### ADB version

Android Debug Bridge version 1.0.41 Version 34.0.4-10411341 Installed as C:\Windows\system32\adb.exe Running on Windows 10.0.19045

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2025-01-06:
Please try to open UADNG from a terminal (PowerShell, cmd prompt, whatever...), then 📋copy-paste the output here

bobcov, 2025-01-09:
I am sorry for the delay. I keep forgetting to test this on the laptop when I get home from work.  It is working on another Windows 10 laptop, so I will need to compare and try to figure out the difference.

bobcov, 2025-01-12:
Hi,
   It works fine on the machine where it did not launch previously. I have no idea what changed. It's really strange, as I tried regular command prompt, powershell, with and without admin rights. Nothing ever happened.  But today, it launched. 
  Now i need to read more about how to use it because there are things on the list that look as if other things will break if I remove them.
 
-Bob

Rudxain, 2025-01-12:
> today, it launched

Maybe Windows updated?

> Now i need to read more about how to use it because there are things on the list that look as if other things will break if I remove them.

Yeah. I also recommend [the FAQ](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ)

bobcov, 2025-01-12:
Hi,
   The last update I see is Feb 15 2024.  I turn off updates.  I will go over the FAQ

closed 2025-01-12 by Rudxain