Universal-Debloater-Alliance/universal-android-debloater-next-generation #775: bug(scope): App crashes after selecting checkboxes

Stikulzon, 2025-01-03:
### Describe the bug

https://youtu.be/PMeHEolgHTk
Log doesn't provide the crash info

### Expected behavior

To not crash

### Operating System

Windows

### Distro version

Win11 Pro

### Provide logs

<details><summary>Log</summary>

```
2025-01-03 04:29:29 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:29:29 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:29:29 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:29:29 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:29:43 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:29:43 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:29:43 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:29:43 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:29:47 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:29:47 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2025-01-03 04:29:48 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:29:48 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:29:48 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:29:53 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:29:53 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2025-01-03 04:29:54 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:29:54 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:29:54 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:29:58 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:29:58 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2025-01-03 04:29:58 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:29:58 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:29:58 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:30:02 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:30:02 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2025-01-03 04:30:03 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:30:03 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:30:03 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-01-03 04:30:06 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-01-03 04:30:06 DEBUG [src\core\update.rs:204] Failed to check UAD-ng update
2025-01-03 04:30:07 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-01-03 04:30:07 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 33 | DEVICE: OnePlus CPH2493
2025-01-03 04:30:07 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------

```
</details>

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Stikulzon, 2025-01-03:
Works fine on Arch

Rudxain, 2025-01-04:
That's odd. Can you please open it from a terminal? That way we can see the `panic` message (it might be a segmentation-fault, considering the OS-specific nature)

Stikulzon, 2025-01-04:
> That's odd. Can you please open it from a terminal? That way we can see the `panic` message (it might be a segmentation-fault, considering the OS-specific nature)

Got: 
```
thread 'main' panicked at src\gui\views\list.rs:248:61:
index out of bounds: the len is 21 but the index is 201
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Rudxain, 2025-01-04:
Assuming you're using the latest stable UADNG, this is the offending line:

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/4b12d034d1046401e6ebfcb3ae581d0b0cce5e4e/src/gui/views/list.rs#L248

Which implies this shouldn't be OS-specific.

I'll try to further inspect/debug the code later. I can't do it right now

Ailen01, 2025-01-22:
I get the exact same error on Arch. It crashes only when I connect my Motorola Edge 40 Pro. When I use it with a Xiaomi Pad 5 Pro, everything works fine.

Rudxain, 2025-01-22:
I tried tracking the variable assignments, but couldn't figure out where `u.index` is set to an out-of-bounds value (assuming `i_package` isn't also OOB).

@Universal-Debloater-Alliance/developers I suggest adding assertions (1 for each index), or ask users to run UAD-NG on a debugger

potchy, 2025-03-10:
I have the same issue. My phone also has work profile enabled. Unticking the checkbox @ghost mentioned worked, but when I click "Uninstall" in any app, it crashes again. :(

blubbsy, 2025-03-16:
i can confirm @potchy s crash, i have it too with 1.1.2. nearly same phone (Poco F6 Pro) and with a work profile, though i'm trying to disable the apps.

NithinV404, 2025-03-25:
i had the same issue but once i disabled the secure folder on my motorola it worked fine

Xon2, 2025-05-15:
On Motorla G 75 disabling (actually deleting) the ''secure folder'' stopped UDA crashing when uninstalling apps as well. Good find, thanks.

theRoboxx, 2025-05-26:
Same issue here on a Poco f3 with work profile enabled. 
```
thread 'main' panicked at src/gui/views/list.rs:959:43:
index out of bounds: the len is 0 but the index is 15
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted (core dumped)
```
Does #972 fixes the uninstall/disable crash as well or just the selection crash?

closed 2025-06-06 by AnonymousWP