Universal-Debloater-Alliance/universal-android-debloater-next-generation #776: feat(scope): Show App Names next to Package Names for better identification

BI0NIC0, 2025-01-03:
### Describe the feature you want

**Current Behavior:**
UAD currently displays apps using their package names (e.g., com.google.android.apps.maps). While accurate, this can be unintuitive for users, who might find it difficult to quickly identify apps.

**Proposed Enhancement:**
Introduce a second column in the app list to display human-readable app names (e.g., "Google Maps") alongside package names. This enhancement could be implemented by extracting app names from APK metadata or system resources (e.g., using PackageManager).

**Benefits:**
- Enhanced user-friendliness by providing recognizable app names.
- Reduced risk of accidental uninstallation of needed apps.
- Improved accessibility for non-technical users.

What do you think about?

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2025-01-04:
Cross-repo dupe of 0x192/universal-android-debloater#59. But thanks for reminding it! We should track progress here

AnonymousWP, 2025-01-06:
Where are the checkboxes you had to check? 😄