Universal-Debloater-Alliance/universal-android-debloater-next-generation #782: pkg(com.qualcomm.qti.powersavemode): Uninstalling this package made my Xiaomi very slow

JoaoHamerski, 2025-01-06:
### Describe the problem

I uninstalled many packages (52 total), my phone got really slow, then I decided to reboot the phone and test it.

It didn't help, it's was very slow even for opening apps who are in the background.

Firstly I thought it was `com.qualcomm.qti.performancemode` because it seems more obvious by the name, then I decided to reinstall `com.qualcomm.qti.powersavemode` and my phone got back to normal.

If this package was really the cause, it might be good a warning or even remove from recommended list.

Can anyone try to replicate it in a Xiaomi phone (especially Poco)? 

- Xiaomi HyperOS 1.0.7.0.ULJMIXM Android 14
- Poco F4 GT

### Provide logs

2025-01-06 17:52:08 ERROR [src/core/config.rs:92] Invalid config file: `TOML parse error at line 1, column 1
  |
1 | theme = "Lupin"
  | ^^^^^^^^^^^^^^^
missing field `general`
`
2025-01-06 17:52:08 ERROR [src/core/config.rs:96] Restoring default config file
2025-01-06 17:52:10 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-01-06 17:52:53 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-01-06 17:52:53 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 21121210G
2025-01-06 17:52:53 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2025-01-06 17:53:49 DEBUG [src/gui/views/settings.rs:110] Config change: Settings { general: GeneralSettings { theme: "Dark", expert_mode: false, backup_folder: "/home/joao/.cache/uad/backups" }, device: DeviceSettings { device_id: "d3f31644", disable_mode: false, multi_user_mode: true, backup: BackupSettings { backups: [], selected: None, users: [User { id: 0, index: 0, protected: false }], selected_user: Some(User { id: 0, index: 0, protected: false }), backup_state: "" } }, is_loading: false, modal: None }
2025-01-06 17:53:49 DEBUG [src/core/config.rs:80] config: New device settings saved
2025-01-06 17:53:59 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-01-06 17:53:59 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 21121210G
2025-01-06 17:53:59 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2025-01-06 18:13:28 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.cellbroadcastservice.overlay.miui -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.rkpdapp -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.gms.location.history -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.managedprovisioning.overlay -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.inputsettings.overlay.miui -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 android.autoinstalls.config.Xiaomi.model -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.settings.overlay.miui -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.cellbroadcastreceiver.overlay.common -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.cellbroadcastreceiver.overlay.miui -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.role.notes.enabled -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.overlay.gmssettings -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.systemui.overlay.common -> Success
2025-01-06 18:13:29 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.turbo -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 android.qvaoverlay.common -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.virtualmachine.res -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.providers.partnerbookmarks -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.captiveportallogin.forframework -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.documentsui -> Success
2025-01-06 18:13:30 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.permissioncontroller -> Success
2025-01-06 18:13:31 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.safetycenter.resources -> Success
2025-01-06 18:13:31 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.mainline.adservices -> Success
2025-01-06 18:13:31 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.mi.globalminusscreen -> Success
2025-01-06 18:13:31 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.mi.healthglobal -> Success
2025-01-06 18:13:31 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.cloudbackup -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.cloudservice -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.micloudsync -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.miwallpaper.overlay -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.miwallpaper.overlay.customize -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.phone.carriers.overlay.h3g -> Success
2025-01-06 18:13:32 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.phone.carriers.overlay.vodafone -> Success
2025-01-06 18:13:33 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.phrase -> Success
2025-01-06 18:13:33 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.settings.rro.device.hide.statusbar.overlay -> Success
2025-01-06 18:13:33 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.settings.rro.device.type.overlay -> Success
2025-01-06 18:13:33 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.system.overlay -> Success
2025-01-06 18:13:33 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.touchassistant -> Success
2025-01-06 18:13:34 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.videoplayer -> Success
2025-01-06 18:13:34 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.miui.wallpaper.overlay -> Success
[... 1 lines cut ...]
2025-01-06 18:13:35 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qti.qualcomm.deviceinfo -> Success
2025-01-06 18:13:35 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.location -> Success
2025-01-06 18:13:35 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.devicestatisticsservice -> Success
2025-01-06 18:13:35 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.performancemode -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.powersavemode -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.qcolor -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.remoteSimlockAuth -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.qti.qms.service.trustzoneaccess -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.qualcomm.uimremoteserver -> Success
2025-01-06 18:13:36 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.quicinc.voice.activation -> Success
[... cut ...]
2025-01-06 18:15:40 INFO  [src/core/sync.rs:131] [Recommended] cmd package install-existing --user 0 com.qualcomm.qti.performancemode -> Package com.qualcomm.qti.performancemode installed for user: 0
2025-01-06 18:15:57 INFO  [src/core/sync.rs:131] [Recommended] cmd package install-existing --user 0 com.qualcomm.qti.powersavemode -> Package com.qualcomm.qti.powersavemode installed for user: 0

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2025-01-11:
> ### Describe the problem
> I uninstalled many packages (52 total), my phone got really slow, then I decided to reboot the phone and test it.
> 
> It didn't help, it's was very slow even for opening apps who are in the background.
> 
> Firstly I thought it was `com.qualcomm.qti.performancemode` because it seems more obvious by the name, then I decided to reinstall `com.qualcomm.qti.powersavemode` and my phone got back to normal.
> 
> If this package was really the cause, it might be good a warning or even remove from recommended list.
> 
> The current description for this package is:
> 
> > It have hidden power saving modes.
> > Users cant use it
> 
> Not clear enough to know if it really affects the performance as I stated.
> 
> Can anyone try to replicate it in a Xiaomi phone (especially Poco)?
> 
> * Xiaomi HyperOS 1.0.7.0.ULJMIXM Android 14
> * Poco F4 GT
> 
> ### You have a solution?
> _No response_
> 
> ### Provide logs
> 2025-01-06 17:52:08 ERROR [src/core/config.rs:92] Invalid config file: `TOML parse error at line 1, column 1 | 1 | theme = "Lupin" | ^^^^^^^^^^^^^^^ missing field `general` ` 2025-01-06 17:52:08 ERROR [src/core/config.rs:96] Restoring default config file 2025-01-06 17:52:10 DEBUG [src/core/update.rs:183] Checking for UAD-ng update 2025-01-06 17:52:53 INFO [src/gui/views/list.rs:160] ----------------------------------------------------------------- 2025-01-06 17:52:53 INFO [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 21121210G 2025-01-06 17:52:53 INFO [src/gui/views/list.rs:165] ----------------------------------------------------------------- 2025-01-06 17:53:49 DEBUG [src/gui/views/settings.rs:110] Config change: Settings { general: GeneralSettings { theme: "Dark", expert_mode: false, backup_folder: "/home/joao/.cache/uad/backups" }, device: DeviceSettings { device_id: "d3f31644", disable_mode: false, multi_user_mode: true, backup: BackupSettings { backups: [], selected: None, users: [User { id: 0, index: 0, protected: false }], selected_user: Some(User { id: 0, index: 0, protected: false }), backup_state: "" } }, is_loading: false, modal: None } 2025-01-06 17:53:49 DEBUG [src/core/config.rs:80] config: New device settings saved 2025-01-06 17:53:59 INFO [src/gui/views/list.rs:160] ----------------------------------------------------------------- 2025-01-06 17:53:59 INFO [src/gui/views/list.rs:161] ANDROID_SDK: 34 | DEVICE: POCO 21121210G 2025-01-06 17:53:59 INFO [src/gui/views/list.rs:165] ----------------------------------------------------------------- 2025-01-06 18:13:28 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.cellbroadcastservice.overlay.miui -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.rkpdapp -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.gms.location.history -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.managedprovisioning.overlay -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.inputsettings.overlay.miui -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 android.autoinstalls.config.Xiaomi.model -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.settings.overlay.miui -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.cellbroadcastreceiver.overlay.common -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.cellbroadcastreceiver.overlay.miui -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.role.notes.enabled -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.overlay.gmssettings -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.systemui.overlay.common -> Success 2025-01-06 18:13:29 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.turbo -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 android.qvaoverlay.common -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.virtualmachine.res -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.providers.partnerbookmarks -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.captiveportallogin.forframework -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.documentsui -> Success 2025-01-06 18:13:30 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.overlay.modules.permissioncontroller -> Success 2025-01-06 18:13:31 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.safetycenter.resources -> Success 2025-01-06 18:13:31 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.mainline.adservices -> Success 2025-01-06 18:13:31 INFO [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.mi.globalminusscreen -> Success 2025-01- [... cut ...] e installed for user: 0 2025-01-06 18:15:57 INFO [src/core/sync.rs:131] [Recommended] cmd package install-existing --user 0 com.qualcomm.qti.powersavemode -> Package com.qualcomm.qti.powersavemode installed for user: 0
> 
> ### Acknowledgements
> * [x]  This issue is not a duplicate of an existing one.
> * [x]  I have chosen an appropriate title.
> * [x]  All requested information has been provided properly.

Could you confirm this in some way? Then we can change it in the documentation. Thanks.

JoaoHamerski, 2025-01-11:
@AnonymousWP Currently I'm back to MIUI 14, not sure if this package was the cause. I was hoping someone else with Xiaomi/Poco could try to replicate and see if it's indeed true, but just throwing it here in case someone have a similar issue.

Currently testing on MIUI 14, it didn't seem to affect anything (the issue was on HyperOS 1)

AnonymousWP, 2025-02-04:
I see. Let me know if you found something worth mentioning.