Universal-Debloater-Alliance/universal-android-debloater-next-generation #784: pkg(poco):

spectrMeltdown, 2025-01-08:
### List your packages

- Package name: com.amazon.appmanager
- List: Misc
- Removal: Recommended
- Description: Maybe it's used for Amazon related apps? Safely removed it myself.

- Package name: com.mi.global.pocostore
- List: OEM
- Removal: Recommended
- Description: Platform for users to browse and purchase Xiaomi and POCO products. If you don't shop using this or use its customer support, its safe to remove.

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2025-01-08:
Could you please open a PR yourself? That'd help us.

spectrMeltdown, 2025-01-10:
okay

schbrns, 2025-04-23:
Thank you for your contribution.
It seems that both already exist and have more detailed descriptions.
Please feel free to re-open if needed.

closed 2025-04-23 by schbrns