Universal-Debloater-Alliance/universal-android-debloater-next-generation #810: pkg(scope):

TheBlunius, 2025-01-24:
### List your packages

- Package name: com.mi.globalbrowser
- Current list: oem
  - Proposed list: oem
- Current removal: Recommended
  - Proposed removal: Recommended
- Current description: Mi Browser
-Privacy nightmare. You should really use something else.
https://www.xda-developers.com/xiaomi-mi-web-browser-pro-mint-collecting-data/incognito-mode/

Note: Since MIUI 12, you can no longer uninstall this app. Disabling it still works fine.
  - Proposed description: Mi Browser
-Privacy nightmare. You should really use something else.
https://www.xda-developers.com/xiaomi-mi-web-browser-pro-mint-collecting-data/incognito-mode/

Note: Since MIUI 12, you can no longer uninstall this app. Disabling it still works fine.
Note: Uninstalling the app works fine in HyperOS 2 (android 15).

### Acknowledgements

- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

TheBlunius, 2025-01-24:
(Tested on Poco X6 5g)

schbrns, 2025-04-23:
Thank you for your contribution.

I'm of the mind that the description is accurate in that, it can still be uninstalled on anything that isn't `MIUI 12 and later`.

Feel free to reopen if needed. Thanks.

closed 2025-04-23 by schbrns