Universal-Debloater-Alliance/universal-android-debloater-next-generation #813: bug(scope): the app CONSTANTLY crashes.

zanezonair, 2025-01-26:
### Describe the bug

the app CONSTANTLY crashes making it hard to work with the thing.

### Expected behavior

do anything. even scrolling.
it just crashed when it wants to.

### Operating System

Windows

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [x] This issue is not a duplicate of an existing bug report.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

AnonymousWP, 2025-01-26:
You need to show logs, otherwise we can't do anything.

zanezonair, 2025-01-26:
> You need to post logs, otherwise we can't do anything.

Sorry, was just mad due to it crashing to often. Managed to get what I needed done with it though.

I'll send any data and update the post with any data needed within 48 hours.

zanezonair, 2025-01-26:
Or I close it by accident like an idiot lol

zanezonair, 2025-01-27:
[UAD_20250125.log](https://github.com/user-attachments/files/18553279/UAD_20250125.log)
[UAD_20250126.log](https://github.com/user-attachments/files/18553280/UAD_20250126.log)

there ya go.

hope these help.

Rudxain, 2025-01-29:
The problem with logs (in general, not just these) is that they never include unexpected `panic` messages. We'll have to update the issue template to recommend running the app from a terminal

zanezonair, 2025-01-29:
Long as I can get a laymans terms guide on how to do that then im all for it.

The cable I use doesn't have issues when transferring data and all so I know it isn't my cable causing the issue.

Rudxain, 2025-01-30:
It's quite easy! You can choose between "Command prompt" or PowerShell. I'd recommend cmdp because PS is bloated and slow (**extremely** slow). To avoid the need to `cd` ("select a directory"), you should open the terminal directly on the same directory as UADNG (I remember it was `shift + right-click` to show the context-menu). The problem is that PS is the default shell+terminal combo, so you might not see the option to open cmdp from the context-menu, that's fine just open PS. Anyways, once the shell has loaded, run `.\
uad-ng-windows.exe` (assuming you haven't renamed the file)

zanezonair, 2025-02-21:
> IDK how to fix this. Seems like an issue with `wpgu` itself. It might be solved if we upgrade Iced

Mind giving me a layman version of this?

And thanks for letting me know, I'll just deal with the crashes like I have been.

Rudxain, 2025-02-21:
> Mind giving me a layman version of this?

`wgpu` is basically the "bridge" of communication between Iced (the GUI framework we use) and your CPU or GPU (integrated or dedicated), for drawing/rendering stuff. UADNG uses Iced v0.12, but [there's a newer version](https://github.com/iced-rs/iced/releases/tag/0.13.0) we can't use yet, because it requires rewriting major parts of our codebase.

> And thanks for letting me know, I'll just deal with the crashes like I have been.

Wait, does your PC have a dedicated graphics card? Maybe this is related to #850

zanezonair, 2025-02-22:
> `wgpu` is basically the "bridge" of communication between Iced (the GUI framework we use) and your CPU or GPU (integrated or dedicated), for drawing/rendering stuff. UADNG uses Iced v0.12, but [...]

CPU: AMD Ryzen 7 5700X
GPU: AMD RX 6700XT

Rudxain, 2025-02-22:
Please try setting the `WGPU_BACKEND` [environment-variable](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/set_1) to `gl` before opening UADNG

zanezonair, 2025-02-23:
> Please try setting the `WGPU_BACKEND` [environment-variable](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/set_1) to `gl` before opening UADNG

uh... i did my best to read all that and it fully went over my head.
please just give me the command i need to use.

user583953, 2025-02-23:
Open `cmd` (should display in the title "Command Prompt", will not work in PowerShell) and execute:

```
set WGPU_BACKEND="GL"
```

After that in the same terminal window:

```
.\uad-ng-windows.exe
```

zanezonair, 2025-02-23:
C:\Users\zanezonair\Desktop\Pictures\Other Items\platform-tools>2025-02-23 05:06:01 WARN  [C:\Users\runneradmin/.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.4\src\instance.rs:1182] unknown backend string '"gl"'
2025-02-23 05:06:01 WARN  [C:\Users\runneradmin/.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.4\src\instance.rs:1189] no valid backend strings found!
thread 'main' panicked at C:\Users\runneradmin/.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-core-0.19.4\src\instance.rs:521:39:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

gave this massive error.

> Open `cmd` (should display in the title "Command Prompt", will not work in PowerShell) and execute:
> 
> ```
> set WGPU_BACKEND="gl"
> ```
> 
> After that in the same terminal window:
> 
> ```
> .\uad-ng-windows.exe
> ```

thank you.

user583953, 2025-02-23:
Sorry, my bad, apparently windows treats ` " ` as plain text. Try the same but:

```
set WGPU_BACKEND=gl
```

zanezonair, 2025-02-23:
> Sorry, my bad, apparently windows treats `"` as plain text. Try the same but:
> 
> ```
> set WGPU_BACKEND=gl
> ```

yep this fixed it. 
thank you both.

closed 2025-02-23 by AnonymousWP

Rudxain, 2025-02-23:
BTW, if you want the variable to persist (no need to run command every time), you should use `setx`

zanezonair, 2025-02-23:
> BTW, if you want the variable to persist (no need to run command every time), you should use `setx` or the built-in Windows GUI

How do I do that?

Rudxain, 2025-02-24:
```cmd
setx WGPU_BACKEND gl
```

https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/setx

Rudxain, 2025-02-24:
```cmd
setx WGPU_BACKEND gl
```

I omitted `/m` because I assume you haven't installed the app for all users

[Official docs](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/setx)