Universal-Debloater-Alliance/universal-android-debloater-next-generation #836: pkg(scope): com.android.unisoc.telephony.server

kyoforkshomebrews, 2025-02-06:
### List your packages

- Package name: com.android.unisoc.telephony.server
- Current list: Oem
  - Proposed list: Oem
- Current removal: Recommended
  - Proposed removal: Expert
- Current description: UICC stands for Universal Integrated Circuit Card.\nIt is a the physical and logical platform for the USIM and may contain additional USIMs and other applications.\n(U)SIM is an application on the UICC.\nOnly for China.
  - Proposed description: Unisoc 4G and 5G drivers. Removal will break incoming calls. https://xdaforums.com/t/oukitel-wp23-pro-unlocking-bootloader-rooting-gsi.4642483/#post-89239693

### Acknowledgements

- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

ThatGreenBlur, 2025-03-01:
there is no action on this for 3 weeks now? this seems too major of a issue to be left behind

AnonymousWP, 2025-03-01:
Sorry for that. Due to personal reasons, I haven't really been able to work on it, and only me and another team member usually file PRs. This remains a voluntary open-source project where we invest our valuable free time in. I'll directly open a PR as it seems more urgent than I thought.

ThatGreenBlur, 2025-03-01:
> Sorry for that. Due to personal reasons, I haven't really been able to work on it, and only me and another team member usually file PRs. This remains a voluntary open-source project where we invest our valuable free time in. I'll directly open a PR as it seems more urgent than I thought.

@AnonymousWP thank you for your immediate attention on this! dont forget to connect the description with com.unisoc.phone aswell as it seems to be related with this issue, once again thanks!

AnonymousWP, 2025-03-01:
> @AnonymousWP thank you for your immediate attention on this! dont forget to connect the description with com.unisoc.phone aswell as it seems to be related with this issue, once again thanks!

Good like this? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/877/commits/8aede76b862245e2fd8ea857c945375b0e4bbb84

ThatGreenBlur, 2025-03-01:
> > > Sorry for that. Due to personal reasons, I haven't really been able to work on it, and only me and another team member usually file PRs. This remains a voluntary open-source project where we invest our valuable free time in. I'll directly open a PR as it seems more urgent than I thought.
> > 
> > 
> > [@AnonymousWP](https://github.com/AnonymousWP) thank you for your immediate attention on this! dont forget to connect the description with com.unisoc.phone aswell as it seems to be related with this issue, once again thanks!
> 
> Good like this? [8aede76](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/commit/8aede76b862245e2fd8ea857c945375b0e4bbb84)

yup! great change

ThatGreenBlur, 2025-03-01:
@AnonymousWP wait a second, the source also indicates com.unisoc.networksliceserver tho i dont think we have an entry for that? maybe you should add it, my bad

AnonymousWP, 2025-03-01:
> @AnonymousWP wait a second, the source also indicates com.unisoc.networksliceserver tho i dont think we have an entry for that? maybe you should add it, my bad

Create a new issue for it. If it's not listed, people won't see it in the app, and thus can't remove it.

closed 2025-03-18 by AnonymousWP