Universal-Debloater-Alliance/universal-android-debloater-next-generation #843: pkg(scope): Fingerprint sensor no longer working after removing all recommended packages on LG Stylo 3 Plus

Lost-Entrepreneur439, 2025-02-08:
### Describe the problem

After uninstalling all recommended packages on the LG Stylo 3 Plus (Android 7.0), the device's fingerprint sensor has stopped working. After initially adding the fingerprint, it'll detect the finger on the sensor but not recognize the finger, and after rebooting it doesn't detect the finger on the sensor at all.

### Provide logs

2025-02-07 21:02:54 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-02-07 21:02:57 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-02-07 21:02:57 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 24 | DEVICE: lge LG-M470
2025-02-07 21:02:57 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2025-02-07 22:05:19 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 492, sub_id: 0 }
2025-02-07 22:18:57 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 757, sub_id: 0 }
2025-02-07 22:54:45 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1132, sub_id: 0 }
2025-02-07 23:28:39 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1682, sub_id: 0 }
2025-02-07 23:28:46 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1706, sub_id: 0 }
2025-02-07 23:28:54 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1714, sub_id: 0 }
2025-02-07 23:28:57 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1721, sub_id: 0 }
2025-02-07 23:29:02 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1737, sub_id: 0 }
2025-02-07 23:29:04 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/calloop-0.12.4/src/loop_logic.rs:529] [calloop] Received an event for non-existence source: TokenInner { id: 3, version: 1742, sub_id: 0 }
2025-02-07 23:31:38 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-02-07 23:31:41 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-02-07 23:31:41 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 24 | DEVICE: lge LG-M470
2025-02-07 23:31:41 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2025-02-07 23:32:08 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.facebook.katana -> Success
2025-02-07 23:32:08 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.facebook.appmanager -> Success
2025-02-07 23:32:08 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.providers.partnerbookmarks -> Success
2025-02-07 23:32:08 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 ca.bell.wt.android.tunesappswidget -> Success
2025-02-07 23:32:09 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.cloudprint -> Success
2025-02-07 23:32:09 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.calllogbackup -> Success
2025-02-07 23:32:09 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.facebook.system -> Success
2025-02-07 23:32:09 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.docs -> Success
2025-02-07 23:32:10 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.egg -> Success
2025-02-07 23:32:10 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.android.bluetoothmidiservice -> Success
2025-02-07 23:32:10 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.evernote -> Success
2025-02-07 23:32:10 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.docs.editors.docs -> Success
2025-02-07 23:32:11 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.docs.editors.sheets -> Success
2025-02-07 23:32:11 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.docs.editors.slides -> Success
2025-02-07 23:32:12 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.maps -> Success
2025-02-07 23:32:12 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.photos -> Success
2025-02-07 23:32:13 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.apps.tachyon -> Success
2025-02-07 23:32:13 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.calculator -> Success
2025-02-07 23:32:14 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.feedback -> Success
2025-02-07 23:32:14 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.gm -> Success
2025-02-07 23:32:14 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.marvin.talkback -> Success
2025-02-07 23:32:14 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.music -> Success
2025-02-07 23:32:15 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.onetimeinitializer -> Success
2025-02-07 23:32:15 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.videos -> Success
2025-02-07 23:32:15 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.google.android.youtube -> Success
2025-02-07 23:32:16 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.instagram.android -> Success
2025-02-07 23:32:16 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.appbox.client -> Success
2025-02-07 23:32:16 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.bnr -> Success
2025-02-07 23:32:17 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.bnr.launcher -> Success
2025-02-07 23:32:17 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.cic.eden.service -> Success
2025-02-07 23:32:18 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.drmservice -> Success
2025-02-07 23:32:18 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.easyhome -> Success
2025-02-07 23:32:18 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.effect -> Success
2025-02-07 23:32:18 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.eltest -> Success
2025-02-07 23:32:19 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.email -> Success
2025-02-07 23:32:19 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.fmradio -> Success
2025-02-07 23:32:19 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.gcuv -> Success
2025-02-07 23:32:20 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.gnss.airtest -> Success
2025-02-07 23:32:20 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.gnsslogcat -> Success
2025-02-07 23:32:20 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.gnsspostest -> Success
2025-02-07 23:32:21 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.gnsstest -> Success
2025-02-07 23:32:21 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.homeselector -> Success
2025-02-07 23:32:21 INFO  [src/core/sync.rs:131] [Recommended] pm uninstall --user 0 com.lge.ia.task.incalagent -> Success
[... cut ...]
2025-02-07 23:52:10 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-02-07 23:52:10 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 24 | DEVICE: lge LG-M470
2025-02-07 23:52:10 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2025-02-07 23:57:20 DEBUG [src/core/update.rs:183] Checking for UAD-ng update

### Acknowledgements

- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

lawson85, 2025-03-19:
It might be the DRM packages but I do not have an LG device. You will just have to restore packages until you get results. Things like Facebook and Gmail do not need to be restored

Muriilow, 2025-11-08:
looking at the packages none of them should be responsible to this problem. @Lost-Entrepreneur439 Did you find a way to fix it? Could you update us?

Lost-Entrepreneur439, 2025-11-14:
restored one to stock and removed all recommended apps, and now it works. strange. will close this issue since its seemingly fixed.