Universal-Debloater-Alliance/universal-android-debloater-next-generation #847: pkg(scope): com.facebook.katana is not removed

starbrights, 2025-02-09:
### List your packages

- Package name: com.facebook.katana
- List: misc
- Removal: 
- Description: 

Sorry, I am new here. Maybe there is some misunderstanding. I removed on Samsung S22  all recommended packages. But after booting there is still a facebook icon and the com.facebook.katana. 
Not sure what it is for, but should be removed too, right?

### Acknowledgements

- [ ] I removed all those packages on my phone. If not, why? Explain in the issue.
- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

ScharadaValcta, 2025-02-09:
the package exists 

"com.facebook.katana": {
    "list": "Misc",
    "description": "Facebook app (https://play.google.com/store/apps/details?id=com.facebook.katana)\n",
    "dependencies": [],
    "neededBy": [],
    "labels": [],
    "removal": "Recommended"
  },

and yes it should be getting removed maybe you dont have the newest version / uad_list.json file

AnonymousWP, 2025-02-10:
Did you uninstall or disable?

closed 2025-11-14 by Muriilow