Universal-Debloater-Alliance/universal-android-debloater-next-generation #848: bug(scope): Wayland crash when runnning on Vulkan, switching backend to GL fixes the issuse.

user583953, 2025-02-12:
### Describe the bug

Running using either of:

`./uad-ng-noselfupdate-linux`
`WGPU_BACKEND=vulkan ./uad-ng-noselfupdate-linux`

results in a crash with error (see log).

Runs normally using: `WGPU_BACKEND=gl ./uad-ng-noselfupdate-linux`

### Expected behavior

No crash.

### You have a solution?

Instead of using vulkan backend, use gl.

https://github.com/gfx-rs/wgpu?tab=readme-ov-file#environment-variables

Run with: `WGPU_BACKEND=gl ./uad-ng-noselfupdate-linux`.

### Operating System

Linux

### Distro version

Fedora 41

### Window System + Manager

interface: 'wl_compositor', version:  6, name:  1 | KWin version: 6.2.5

### Desktop Environment

KDE Plasma 6.2.5

### Provide logs

<details><summary>Log (copy paste from terminal)</summary>

```
[destroyed object]: error 7: importing the supplied dmabufs failed
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
2025-02-12 16:54:25 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 2025-02-12 16:54:25 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR

thread 'Protocol error main7' panicked at  on object /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.4/src/device/global.rs@:01859: :
21:
Protocol error internal error: entered unreachable code: Fallback system failed to choose present mode. This is a bug. Mode: AutoVsync, Options: []7
 on object note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
@0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0: 
Protocol error 7 on object @0:
```
</details>

<details><summary>Log (UAD_20250212.log)</summary>

```
2025-02-12 16:54:25 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1824] get_physical_device_surface_present_modes: ERROR_SURFACE_LOST_KHR
2025-02-12 16:54:25 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/adapter.rs:1839] get_physical_device_surface_formats: ERROR_SURFACE_LOST_KHR

```
</details>

### Acknowledgements

- [x] This issue is not a duplicate of an existing bug report.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

AnonymousWP, 2025-02-13:
Hey, thanks, we're aware of this issue. However, we're currently struggling with fixing it. Could you please help us out? See https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/794/files.

AnonymousWP, 2025-02-14:
Thanks a lot! I'll assign our developers to check your PR.

PR #850 (2025-02-18): fix: force `wgpu` to use dedicated/discrete graphics-card, to avoid Vulkan issues
I've spent like an hour looking through Iced, and apparently there is no option to pass config to wgpu, it always picks settings from env:

https://docs.rs/iced_renderer/0.12.1/src/iced_renderer/compositor.rs.html#252
https://docs.rs/iced_wgpu/0.12.1/src/iced_wgpu/settings.rs.html#47

Also, for some reason, on my PC setting via WGPU_BACKEND=gl (either in terminal or using set_env) only fixes the issue when compiling with `--release`.

fixes #848
closed 2025-02-18 by AnonymousWP