Universal-Debloater-Alliance/universal-android-debloater-next-generation #849: bug(scope): No apps list on Gingerbread, looks like it doesn't recognize the pm command.

Lost-Entrepreneur439, 2025-02-12:
### Describe the bug

When running the app while connected to a device running Android Gingerbread, nothing shows, and if you set it to show all packages, it shows a bunch of blank packages, then shows a bunch of pm commands, looking like pm didn't recognize the command.

[image]

### Expected behavior

uad-ng to list the apps on the device.

### Operating System

Linux

### Distro version

Debian Trixie

### Window System + Manager

Wayland

### Desktop Environment

KDE Plasma 6.2.5

### ADB version

1.0.41

### Provide logs

<details><summary>Log</summary>

```
2025-02-12 15:36:39 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-02-12 15:36:40 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2025-02-12 15:36:40 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 10 | DEVICE: MOTO XT860
2025-02-12 15:36:40 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------

```
</details>

### Acknowledgements

- [x] This issue is not a duplicate of an existing bug report.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

Rudxain, 2025-02-16:
There's no full support for Android versions below 4.0, but that's a bug nonetheless. The app should detect if the command fails and refuse to show "garbage", but that doesn't happen, for some reason 🤔. Either the failure `pm` doesn't "bubble-up" to adb, or old versions of `pm` return exit-code `0` even if a flag isn't recognized.

Can you please try running this manually?
```sh
adb shell pm list packages -s
```
It seems it doesn't recognize the `-s` flag, so maybe we'll have to add support for [showing all packs](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/592#issuecomment-2576278188)

Lost-Entrepreneur439, 2025-11-02:
Update - Now running on GIngerbread it doesn't even try, uad-ng just crashes.