Universal-Debloater-Alliance/universal-android-debloater-next-generation #851: Is "android" a valid package-name?

Rudxain, 2025-02-18:
#757 introduced a `debug` regression (doesn't effect `release`, yet), which happens because `pm list packages -s` includes the line "package:android", which breaks the assumption that all packs must be valid [`PackageId`](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/52b78a99b49c2de52858f6a96782f17fb8a94c4c/src/core/adb.rs#L181-L198)s

[The problematic pack is part of the list](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/52b78a99b49c2de52858f6a96782f17fb8a94c4c/resources/assets/uad_lists.json#L29722-L29729), so this is a confusing situation (at least for me)

So what should we do about it?
- Strip the pack from `pm` output (runtime), and remove it from the `uad_lists` (static), to pretend it doesn't exist
- Consider "android" a special-case for `PackageId`, making its invariant more complicated
- Stop assuming `pm` prints valid package-IDs, implying [`list_packages_sys_parsed`](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/52b78a99b49c2de52858f6a96782f17fb8a94c4c/src/core/adb.rs#L283-L293) shouldn't even exist
- Something else?

AnonymousWP, 2025-02-18:
Do you believe this should be fixed in our upcoming release (1.1.1)?

adhirajsinghchauhan, 2025-02-18:
It's been a while since I've touched the codebase. Why are we validating package names? Is there an instance where we operate on the output of `pm list packages -s` directly? Are we throwing them around to intermediate ADB APIs? If not, the JSON list is ultimately what's important, that's our source of truth. If a package name does not exist in the list, we do not care about it. So, we should skip the overhead of an unnecessary regex validation step.

Rudxain, 2025-02-18:
> Do you believe this should be fixed in our upcoming release (1.1.1)?

On my local repo, I've just [commented this](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/52b78a99b49c2de52858f6a96782f17fb8a94c4c/src/core/adb.rs#L271-L272), so that's the hotfix I was planning.

> Why are we validating package names?

As of now, it's only validated in `debug` mode. I wanted to ensure the regex is robust enough to be used in CI.

> Is there an instance where we operate on the output of `pm list packages -s` directly?

Nope. As of now, the code doesn't care about the invariant of `PackageId`; but _it should_, for the Serde test:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/52b78a99b49c2de52858f6a96782f17fb8a94c4c/src/core/uad_lists.rs#L243

It would only be important when we add the [`disable` and `uninstall` methods](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/757#issuecomment-2620475667), but even those would typically get their args from the the list-file and `pm`, so the validation isn't important.

The only use-case that seems helpful at runtime, is validating **user backups**. UADNG could show a warning if any pack-name is invalid.

> If not, the JSON list is ultimately what's important, that's our source of truth.

I agree. For now, CI is the only thing that should be prioritized