Universal-Debloater-Alliance/universal-android-debloater-next-generation #853: build(ci): build artifacts when Rust-code is updated

AnonymousWP, 2025-02-18:
> Because the PR is from a fork, I can't kick off the 'Build artifacts' Action

What if we modify the CI itself to be allowed to generate artifacts from PRs? That's [what Termux does](https://github.com/termux/termux-app/actions/workflows/debug_build.yml)

_Originally posted by @Rudxain in https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/850#discussion_r1958277784_

Rudxain, 2025-02-19:
To clarify, about the "heavy" concerns, it doesn't need to run automatically, only manually

AnonymousWP, 2025-02-19:
> To clarify, about the "heavy" concerns, it doesn't need to run automatically, only manually

So you mean it should be possible to manually kick off building artifacts? Cause that what you seem to imply, but that's already possible: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/.github/workflows/build_artifacts.yml#L3-L4. If that's not what you mean, feel free to explain.

AnonymousWP, 2025-03-08:
Well, considering we can already build artifacts from our own branches and not forks, I'm considering closing this issue. Any other thoughts?

Rudxain, 2025-03-08:
I guess it can be closed for now. There's not a "high enough" need for this (right now). And it seems the artifacts can still be generated on the fork side, so we aren't "missing out" on anything

closed 2025-03-08 by Rudxain