Universal-Debloater-Alliance/universal-android-debloater-next-generation #855: bug(scope): Unable to run on windows 11

pa0808, 2025-02-19:
### Describe the bug

I am unable to open the app and nothing is showing on the screen. There is just a single pop up of Windows has detected an unsafe app then when i click on Run anyway the app does not open.

[image]
[image]

### Expected behavior

The app should open 

### Operating System

Windows

### Distro version

Windows 11 Version 24H2 for x64-based Systems (KB5051987)

### Provide logs

<details><summary>Log</summary>

```
Please replace this line by the log-file contents
```
</details>

### Acknowledgements

- [x] This issue is not a duplicate of an existing bug report.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

AnonymousWP, 2025-02-19:
Please show logs, otherwise we can't do anything.

y4ee, 2025-02-25:
@pa0808 

1. Download this file and unzip it: [platform-tools-latest-windows.zip](https://dl.google.com/android/repository/platform-tools-latest-windows.zip).

2. Place the ```uadng.exe``` file inside the unzipped folder and run the executable from there.

> [!NOTE]
> It seems like you might have tried to run the uadng.exe directly after downloading it, but it doesn’t work that way.

Furthermore, check the [guide](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Getting-started)