Universal-Debloater-Alliance/universal-android-debloater-next-generation #865: feat(scope): Need a companion app on device to restore packages if needed

starbrights, 2025-02-22:
### Describe the feature you want

The programm helps a lot to debloat. But in case you go to far and you are not at home (very likely) that it would be required to restore apps without a PC.

BTW: I am wondering what command (adb ?) is used under the hood, and could one see the result (which packages is uninstalled) via adb shell?

### Acknowledgements

- [x] This issue is not a duplicate of an existing feature request.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

AnonymousWP, 2025-03-01:
We're not planning to develop an Android app, there's [Canta](https://github.com/samolego/Canta) that can do the same and uses our list.

starbrights, 2025-03-07:
Thanks. Canta is using Shizuku, that is a drawback.
I found AppManager as a mighty tool.

If you do know: What mechanism is used for debloating? adb ? Do you have more details?

AnonymousWP, 2025-03-08:
> Thanks. Canta is using Shizuku, that is a drawback. I found AppManager as a mighty tool.
> 
> If you do know: What mechanism is used for debloating? adb ? Do you have more details? Found in FAQ:
> 
> ```
> Uninstall: pm uninstall --user <user> <package>
> Restore: cmd package install-existing --user <user> <package>
> Disable: pm disable-user --user <user> <package> + pm am force-stop --user <user> <package> + pm clear --user <user> <package>
> Enable: pm enable <package>
> ```

Yeah, see our Wiki:

- https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/ADB-Reference
- https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#what-are-the-adb-commands-used-by-uad-ng

Does this answer your question?