Universal-Debloater-Alliance/universal-android-debloater-next-generation #870: pkg(scope): com.coloros.ocs.opencapabilityservice

GenTEKS, 2025-02-25:
### List your packages

- Package name: com.coloros.ocs.opencapabilityservice 
- Current list: Recommended/OEM
  - Proposed list: 
- Current removal: Recommended
  - Proposed removal: Advanced
- Current description: Security things and something related to chinese payment
  - Proposed description:  Also used for live alerts by front facing camera/status bar for iPhone-like style music widget (for Spotify), battery charge indicator, flashlight indicator and more.

### Acknowledgements

- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

athnnn, 2025-03-21:
i think its also used for the ProXDR toggle in the photo app

AnoR12345, 2026-09-09:
- I can attest to this issue, com.coloros.ocs.opencapabilityservice  is a dependency for Live Alerts. 
- Uninstalling it leads to the Live Alerts feature plainly not working. 
- Restoring it and rebooting fix the issue. 
- Device: OnePlus Nord 5 (OnePlus CPH2707) 
- Software version: CPH2707_15.0.2.600(EX01B100P01)

AnonymousWP, 2026-09-17:
Can someone create a PR?