Universal-Debloater-Alliance/universal-android-debloater-next-generation #871: docs(wiki): link to script for automatically fetching packages

AnonymousWP, 2025-02-26:
### Discussed in https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/818

<div type='discussions-op-text'>

<sup>Originally posted by **ScharadaValcta** February  2, 2025</sup>
maybe it is possible to create a button like "get apk"
which does e.g.

```
adb shell pm path com.mediatek.callrecorder
package:/system/priv-app/CallRecorderService/CallRecorderService.apk

adb pull /system/priv-app/CallRecorderService/CallRecorderService.apk

mv CallRecorderService.apk "com.mediatek.callrecorder - CallRecorderService.apk"
```</div>

Rudxain, 2025-03-12:
[Done!](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/How-to-contribute/_compare/14d252bed639165f7ded80c260c2cf8877bcd179...584abc3c8e0aa10c97d6e215f466e2362403c85c)

Rudxain, 2025-03-12:
Although a separate issue for the app button seems appropriate. I'll create it now

closed 2025-03-12 by Rudxain