Universal-Debloater-Alliance/universal-android-debloater-next-generation #876: pkg(scope): com.google.android.contactkeys

AlexeiCrystal, 2025-02-28:
### List your packages

- Package name: com.google.android.contactkeys
- List: 
- Removal: Advanced
- Description: 
Android System Key Verifier is a system service for verifying interlocutors using messengers with end-to-end encryption. The app has two focuses: it gives developers the ability to store encryption keys in a single location, and it allows users to verify that they are actually communicating with the expected conversation partner and not the "man in the middle" (an IT term referring to a situation where an attacker intercepts communication between victims and retransmits or modifies it). In practice, this is so that when adding a conversation partner's contact, if desired, you can scan a QR code from their Android System Key Verifier app, and they have to do the same in reverse (code entry instead of QR code scanning is also supported). This links the interlocutors' devices and their interlocutors' contacts. The idea is that if someone later suspects that the conversation has been compromised, chat participants can show their codes (apparently already new) and check their verification - in case of a mismatch, the utility will display a warning.

### Acknowledgements

- [x] I removed all those packages on my phone. If not, why? Explain in the issue.
- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

Muriilow, 2025-10-24:
Here is also a good and simple explanation by Google: https://developers.google.com/android/binary_transparency/google1p/overview

AlexeiCrystal, 2025-10-26:
I noticed that in the Google Contacts app, a new button has appeared next to each contact to verify the encryption keys between you and the other person, apparently for encrypting RCS messages. When you delete or disable this app, the button disappears.