Universal-Debloater-Alliance/universal-android-debloater-next-generation #892: bug(scope): UAD-NG app Failed to update from 1.1.0 to v1.1.1

OlliesGudh, 2025-03-10:
### Describe the bug

App will show "Failed to update to v.1.1.1" on top bar after trying to update version by either button available on menu bar and "About" screen.

On "Apps" page however it shows "Failed to download update" on screen center despite log-file saying that the update has been downloaded.

### Expected behavior

To start update process

### Operating System

Windows

### ADB version

1.1.0

### Provide logs

<details><summary>Log</summary>

```
2025-03-10 10:43:23 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:43:24 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:43:24 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 35 | DEVICE: Redmi 23028RA60L
2025-03-10 10:43:24 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:43:29 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:43:29 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:43:38 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:43:43 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:43:43 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 35 | DEVICE: Redmi 23028RA60L
2025-03-10 10:43:43 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:43:45 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:43:45 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:43:46 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:43:46 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:43:50 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2025-03-10 10:44:04 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:44:07 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:44:07 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:44:10 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:44:10 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2025-03-10 10:44:10 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:44:10 WARN  [src\core\uad_lists.rs:221] Could not load remote debloat list
2025-03-10 10:44:10 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found
2025-03-10 10:44:11 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:44:11 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:44:13 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:44:13 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2025-03-10 10:44:13 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:44:13 WARN  [src\core\uad_lists.rs:221] Could not load remote debloat list
2025-03-10 10:44:13 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found
2025-03-10 10:44:14 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:44:17 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:44:17 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:44:18 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:44:18 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:45:11 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:45:11 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2025-03-10 10:45:11 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:45:11 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found
2025-03-10 10:49:03 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:49:03 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 10:51:31 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:52:10 ERROR [src\core\utils.rs:99] Can't open the following URL: 
2025-03-10 10:56:31 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:57:37 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 10:57:37 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2025-03-10 10:57:37 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 10:57:38 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found
2025-03-10 10:58:40 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:59:06 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 10:59:08 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 10:59:08 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 11:00:00 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 11:00:50 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 11:00:50 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
2025-03-10 11:00:53 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 11:00:59 DEBUG [src\core\update.rs:183] Checking for UAD-ng update
2025-03-10 11:01:09 INFO  [src\gui\views\list.rs:160] -----------------------------------------------------------------
2025-03-10 11:01:09 INFO  [src\gui\views\list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2025-03-10 11:01:09 INFO  [src\gui\views\list.rs:165] -----------------------------------------------------------------
2025-03-10 11:01:09 WARN  [src\core\uad_lists.rs:221] Could not load remote debloat list
2025-03-10 11:01:09 ERROR [src\gui\views\list.rs:873] AppsView ready but no phone found
2025-03-10 11:01:11 DEBUG [src\gui\mod.rs:280] UAD-ng update has been downloaded!
2025-03-10 11:01:11 ERROR [src\gui\mod.rs:314] Failed to update UAD-ng!
```
</details>

### Acknowledgements

- [x] This issue is not a duplicate of an existing bug report.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

Rudxain, 2025-03-10:
Found the problem:
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/87b5a9a9244aedab690027c9e67c1830a716dd8e/src/core/update.rs#L232-L250
The assets have been renamed from the CI side.

It's impossible to fix this for any current release, but we should fix it for future releases. I propose using the GH API to get the full list of asset names, then select the appropriate file if it contains the OS name. Yes, it's more inefficient, but more robust

AnonymousWP, 2025-03-10:
> Found the problem:
>
> https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/87b5a9a9244aedab690027c9e67c1830a716dd8e/src/core/update.rs#L232-L250
>
>
> The [...]

Can't we publish a hot-fix? e.g. 1.1.2. Or wouldn't that work?

Rudxain, 2025-03-10:
> Can't we publish a hot-fix? e.g. 1.1.2

Yes! What I meant is that we can't (and shouldn't, to avoid confusion) "rewrite history" (`git` tags and GH-releases). Even if we remember to update the names in the code, the older app versions would fail to update. So new logic is necessary to ensure "future old" versions can update even if there's a rename

closed 2025-03-10 by Rudxain

AnonymousWP, 2025-03-10:
Can you test with the new release @OlliesGudh https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/releases/tag/v1.1.2 ?

PR #895 (2025-03-10): hotfix(self-update): differentiate between Intel and Apple Silicon on MacOS and update Windows binary name
Fixes #892