Universal-Debloater-Alliance/universal-android-debloater-next-generation #1004: Packages for Meta Quest 3/2

Abyss-c0re, 2025-06-25:
### List your packages

Package name: com.meta.horizonadidservice
List: OEM
Removal: Recommended
Description: Meta's advertising ID service, used to track and deliver personalized ads in VR environments. Not essential to core functionality.

Package name: com.facebook.horizon
List: OEM
Removal: Recommended
Description: Horizon Worlds social VR app by Meta. A bloated, underused platform with heavy background activity. Not necessary unless you actively use Horizon.

Package name: com.oculus.assistant
List: OEM
Removal: Recommended
Description: Voice assistant for Oculus/Meta Quest. Often runs in the background. Can be removed if voice commands aren't used.

Package name: com.oculus.avatareditor
List: OEM
Removal: Recommended
Description: Used to create and modify Meta avatars. Unnecessary unless you're customizing your avatar often in social VR apps.

Package name: com.oculus.bugreportservice
List: OEM
Removal: Recommended
Description: Internal bug reporting utility. Sends crash and diagnostic data to Meta. Has no impact on daily VR use.

Package name: com.oculus.bugreportuploader
List: OEM
Removal: Recommended
Description: Companion to bugreportservice; uploads logs and diagnostics. Another telemetry element with no user-facing benefit.

Package name: com.oculus.explore
List: OEM
Removal: Recommended
Description: "Explore" hub on Quest, promoting apps, events, and content. Not critical—removing it reduces visual clutter and background activity.

Package name: com.oculus.quickpromotionservice
List: OEM
Removal: Recommended
Description: Delivers popups and promotional messages in the Quest interface. Not required and may affect performance and user experience.

Package name: com.oculus.socialplatform
List: OEM
Removal: Recommended
Description: Enables social VR features like friend invites and chats. Safe to remove if you don’t engage in Meta’s social ecosystem.

Package name: com.oculus.tv
List: OEM
Removal: Recommended
Description: Oculus TV app for consuming Meta’s media and video content. Can be removed if you don’t use it to watch shows or streams in VR.

### Acknowledgements

- [x] I removed all those packages on my phone. If not, why? Explain in the issue.
- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

AnonymousWP, 2025-06-25:
Could you please open a PR yourself? That'd be very helpful, thanks.

Fazzoc, 2026-08-21:
This is something I'll consider looking into and making a PR for. I'll use the descriptions as a starting point and include any new ones that have shown up since this issue was made.