Universal-Debloater-Alliance/universal-android-debloater-next-generation #1069: pkg(scope): Xiaomi 13T half bricked after UAD recommended list

BFW9925, 2025-08-27:
### List your packages

- Package name: recommended
(im newish to this so pls guide me)

I freshly downloaded the latest version of uadng.
Latest update on my xiaomi 13T (maybe that is the issue??)
I used UAD to remove the recommended stuff ( it said it were 144 packages because i have removed some stuff before using the old version of UAD)
the total list of uninstalled packages of the recommended is 234.

I also looked manually at all the 144 packages to see if anything was necessary or if it was something i use, but i did not find any of that.
After doing the latest uninstallment of all the packages the phone seemed fine using the apps that were already open. but after closing they wont open again.
They are clickable, they open and have the startup screen but every app stays in their own respective loading screen.

I tried restoring some packages that said they were useless(maybe they werent) but to no result.
Im hesitant to restart my phone because of potential bootloop problems.

if i need to provide more info im glad to do so.

The (stupid) things i didnt do:
Backup the phone
Disable before uninstalling

### Acknowledgements

- [ ] I removed all those packages on my phone. If not, why? Explain in the issue.
- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

Muriilow, 2025-11-14:
Hi, @BFW9925 can you update us in this one?

BFW9925, 2025-11-22:
Sorry guys, lost my acc info. 

Before restarting the phone i made sure to check all the different files if they had impact on the device but nothing made a difference. In the end i just restarted my phone and everything seems to be working fine.

So I removed all the unnecairy packages from my xiaomi (which is a ton) i also deleted all the google applications and some extra Convience package. The only complaint i do have is that wifi doesnt seem to be working correctly.

My wifi connections keep dropping and the phone switches to 4G/5G and does that twice every minute or so. Ill have to find out what it is because it is really annoying for watching revanced or loading searches on my browser.