Universal-Debloater-Alliance/universal-android-debloater-next-generation #1096: pkg(com.google.android.gms.supervision): Removal of Parental Controls app breaks Google One VPN on Pixel Devices

Scoop2389, 2025-09-25:
### List your packages

- Package name: com.google.android.gms.supervision
- Current list: Google
  - Proposed list: Google
- Current removal: Recommended
  - Proposed removal: Advanced
- Current description: Family Link parental controls\nIt has something to Family Link parental controls.\nIntroduced in android 13.\nhttps://play.google.com/store/apps/details?id=com.google.android.gms.supervision
  - Proposed description: Family Link parental controls\nIt has something to Family Link parental controls.\nIntroduced in android 13.\nhttps://play.google.com/store/apps/details?id=com.google.android.gms.supervision\nWARNING: Causes VPN by Google to stop functioning on Pixel Devices, displays "This account cannot use VPN." error.

### Acknowledgements

- [x] This issue is not a duplicate of an existing one.
- [x] I have chosen an appropriate title.
- [x] All requested information has been provided properly.

Muriilow, 2025-10-24:
Hey! I know Im kinda late, but I wanted to ask one question about this request! 

> WARNING: Causes VPN by Google to stop functioning on Pixel Devices, displays "This account cannot use VPN." error.

Did you remove only "com.google.android.gms.supervision" and then started to have problems with Google VPN, or you disabled a lot of other packges in one "go"? 

I dont see how these two are connected... 

No comments about this issue in Virus Total: https://www.virustotal.com/gui/file/cd32e30b1c22197a0530dee11aed023b3d7055686b9a63b631cafbc16db4cb28/community

Neither in this blog: https://9to5google.com/2023/05/25/google-family-link-parental-controls/

Scoop2389, 2025-10-25:
Hello! No problem about your response timing, that's the reason I opened an issue instead of a PR (so people could check me.) I did only remove that application, while on the Google VPN, which immediately disconnected saying the VPN was temporarily unavailable. I restarted my phone and saw a message stating that the VPN wasn't supported with my Google account. Restoring the package from an Application Manager backup caused the issue to go away after a reboot. 

Personally, my guess is that if your Google account is managed by family link, you can't use the VPN, which it checks by sending a request of some sort to that app. When it's uninstalled, it probably won't work, giving the error. I don't exactly have any evidence for this, but it seems consistent with my experiences with Google parental control apps in the past (restricting a lot of behavior by default that could be "risky" even if the parent can't allow it.)

If you would like, I can redo the process and uninstall the package, but this time taking screenshots making a more detailed list of steps I did.

Scoop2389, 2025-10-25:
Nevermind. Just tried to reproduce it and couldn't. Sorry about that.

Muriilow, 2025-10-25:
> Personally, my guess is that if your Google account is managed by family link, you can't use the VPN, which it checks by sending a request of some sort to that app. When it's uninstalled, it [...]

This logic makes sense, and it wouldnt be a problem to add a little warning. Sometimes an user can disable a lot of packages and find problems and correlate with some package that has nothing to do with it. Thats why I asked. 

> Nevermind. Just tried to reproduce it and couldn't. Sorry about that.

Does that means that the VPN gives no error this time? Or you just wasnt able to reproduce by other reasons?

Muriilow, 2025-10-25:
Yeah, sometimes this happens XD.

Anyway if you find any other problem regarding packages documentations feel free to comment!